ntadmin/DGND3700v2 #34: Feature Request: WOL settings

thinkv, 2016-08-16:
Hi there, would be great to add in settings to configure WOL from WAN.

ntadmin, 2016-08-16:
Hi thinkv,

Thanks for the suggestion, could you clarify what you mean. I think that you are wanting to trigger a Wake-On-Lan action from a machine on your network remotely.

I've not used this technology, but in principle I would assume that if your remote device can send the right LAN packet to the WAN Side, and the router has set up a portmap to the right machine, it should work.

So, if you configure incoming port XXXXXXX on the router to forward to the Wake-On-Lan port on your machine, and then configure your 'send the wake up' machine to send it you <your home ip>:XXXXXX it should get through.

ntadmin, 2016-08-17:
Hi again thinkv,

I've done a bit more reading and it seems that, although my suggestion above would work on many routers, it may not work on Netgear routers. Could you let me know what you have tried, what does and doesn't work and so on so that I can see if I can work round it please?

thinkv, 2016-08-18:
Hi ntadmin,

I should have been clearer. In order to successfully achieve WOL of WAN with this router/modem, you actually need to configure the ARP settings. Most people seem to enable telnet on the device, and configure the routes here.

I have seen a few custom firmwares for other routers that actually integrate the WOL features per device.

There's a lot of talk about it here, i figure it could be integrated into the UI?
https://community.netgear.com/t5/General-WiFi-Routers/WOL-amp-Magic-packet/td-p/24102/page/25

ntadmin, 2016-08-18:
Hi thinkv,

Are you able to test the solution given here: https://community.netgear.com/t5/Nighthawk-WiFi-Routers/Wake-On-Lan-with-R7000/td-p/379989

Telnet can be turned on via other methods, or you can use one of my firmware's and use ssh.

If you can confirm that that works, then we can look at integrating it.

vezza, 2016-10-03:
1) I'm using DGND3700v2_NEWT16B3.img but I'm not able to connect with ssh (connection refused)

2) Hi! I 've tested the solution given here: [https://community.netgear.com/t5/Nighthawk-WiFi-Routers/Wake-On-Lan-with-R7000/td-p/379989](url) but /sbin/arp is missing in my router...

ntadmin, 2016-10-05:
@vezza thanks for giving it a go. Very odd that ash isn't working. Can you telnet in?

I will be releasing an updated beta in the coming days which may resolve any ssh issues as I know there was an instability or two.

Thanks for getting involved, I will also look at the /sbin/arp issue - it may already be resolved in the next beta.

VincentYbarra, 2016-10-05:
I was also trying to help out, SSH never worked when I did try it. 

Can't wait for the new beta :D

vezza, 2016-10-05:
Thanks @ntadmin, Yes I can telnet in but I cannot ssh in. I'll also help as I can... bye!

ntadmin, 2016-10-06:
Cool, thanks both of you, I'll put this one down for B7 assuming we can get it working.

vezza, 2016-10-06:
I'm gonin to test ssh on B6 as soon as you release, and if it doesn't work will open a separate issue

thinkv, 2016-10-07:
thanks ntadmin, can't wait to test. sorry I went awol, didn't have access to the router I was testing with.

ntadmin, 2016-10-07:
I sent it from my phone using the email reply system - guess it didn't get to you. When I am back at my computer I will send the picture.

Thanks,

Neil

ntadmin, 2016-10-07:
Welcome back @thinkv !

All: NEWTB6 now released :-)

ntadmin, 2016-10-09:
Although I can;t test it, now that B6 is (theoretically) out there, I did some work reading the proposed solution and checking some details. I hadn't realised that the arp absence noted above was a refewrence to the firmware on this router. So, with https://github.com/ntadmin/DGND3700v2/commit/add3db1de91b639f8cabaa3ac274c70a9671f1cf I have changed the firmware so that it should be there.

I will release B7 in the next couple of days with only two changes from B6: arp available and more stable adblock httpd server.

ntadmin, 2016-10-10:
B7 released including /usr/sbin arp. So ti should be possible to test the proposed method by hand. Once it works by hand and have a clear method for turning it on and off, we can implement it via the web control panel etc etc.

thinkv, 2016-10-12:
Thanks @ntadmin , it did seem to work, however after some idle time the WOL packet no longer reaches the address, however it does work on the local network, very strange.

Also, how can we revert to the stock firmware? I tried to update to the original file from NG, but it does not replace yours.

ntadmin, 2016-10-12:
Great news - I suspect that we can make it work then. I will do some work on a version with it in ... it will be a few days before I have the time though.

@thinkv, are you using pppoe to a modem? If so, ten unplug the pppoe, wait a couple of minutes or so, and then trigger the firmware change on the web interface.

If not, not sure why it doesn't work!

ntadmin, 2016-10-17:
OK, so the plan for implementation is noted on this web page for reference, here it is:

- Enable to routing to know the IP -> MAC link using arp: `arp -i br0 -s x.x.x.x mm:mm:mm:mm:mm:mm` where x.x.x.x is the WoL destination IP address (not .255) and mm... is
the WoL destination's MAC address
- Set up forwarding of port 9/UDP WAN side to the IP of the device LAN side.

So the web page will need to user to enter:

- Enable / disable option
- IP of device to trigger
- MAC of device to trigger

ntadmin, 2016-10-17:
As an aside, it is worth noting that `arp` is actually deprecated as a command, and that `ip neighbor` should be used instead. However, this will work well for initial testing.

ntadmin, 2016-10-17:
Ok, the git code now contains the first go at turning WOL on and off. Note:

- I haven't had a chance to do any testing of the code, there may still be silly bugs.
- It won't start on boot.
- So, to start it, go to the Wake on LAN web page (Advanced -> Advanced Setup), set up the details, enable, and save.

If the code is correct, on the basis of @thinkv above, it should work, at least for a period of time. If that much works, we know we are on the right path.

ntadmin, 2016-10-19:
OK, the code I have just pushed into the git repo has been 'tested' on my machine and it seems to do what I would expect. However, I don't have a WOL set up to test that it actually works. IT should aslo restart on reboot etc etc.

So, this code is now in alpha, ready to be tested.

vezza, 2016-10-27:
I think that if you do WOL on the LAN side from a router's web pace it isn't necessary to open 7 and 9 porta and do the ARP setting... Just broadcast the magic packet to the lan broadcast address (usually .255). Also on my router br0 is not there,  do someone have it?

ntadmin, 2016-10-27:
Hi @vezza, I think that you are right, but to do that we would have to explicitly catch the WOL packet coming in on the WAN side and then broadcast on the LAN side. I think that 'br0' is 'group1' on this router. If you have a proposed way of doing this I'm interested to hear it ...

vezza, 2016-10-28:
Hi @ntadmin, you are right... I was meaning that there are 2 ways to have a working WOL:
1) WOL packet coming from an outside network (Wake On WAN) - this is the most complex scenario and it  needs to broadcast the WOL Packet to the LAN side;
2) WOL packet generated from the router (LAN side) - that should be a more simple scenario but it is required to have access to the router or  maybe a planned cron job/daemon that send the WOL packet at a specific time;
If we can have 1) it would be the best but if it is not possible it will be pleasant to have at least 2)
I am testing the 1) and I have this problem: The router is crashing when I issue the following command: arp -i group1 -s 192.168.0.4 FF:FF:FF:FF:FF:FF
but works keeps working if i type: arp -i group1 -s 192.168.0.255 FF:FF:FF:FF:FF:FF
I still don't know if the WOL Packet is getting broadcasted from the WAN to the LAN side because I am having issue with the DDNS redirect... Now I'll test with IP and will report later

vezza, 2016-10-28:
So, I tested WOL sending Magic Packet from outside WAN to the router IP, before sending the WOL Packet I opened ports 7 and 9 to LAN IP 192.168.0.4 and executed the following command on the router: arp -i group1 -s 192.168.0.255 FF:FF:FF:FF:FF:FF

Result: The WOL packet didn't reached the LAN, tested on the LAN side it works fine

ntadmin, 2016-10-28:
Thanks for the testing, interesting. The failed arp command is very odd. The WAN test you did I would expect to fail because the arp command is telling the system about .255 whilst the port pass through is looking for .4 ....
I will look into the arp failure when I am next at home (around the first of November) and have access to the router!
I assume that by ff:ff:etc you mean the actual MAC address of the machine?

vezza, 2016-10-28:
Ok, thanks... No I assume ff:ff:etc as a broadcast mac address as specified here [https://www.dd-wrt.com/wiki/index.php/WOL](url)

ntadmin, 2016-11-04:
Just having a quick look at this and I can execute the .4 arp command fine. Is there a device on that number? The process in the reference you provide makes it clear that it must be an unused IP - if (say), the router is using that IP, you are in trouble!

On another, note, thank you for that link - it provides a much neater way of doing it than the previous suggestion. I will have a look at implementing it.

ntadmin, 2016-11-04:
I've justed pushed to the git source updates which mean that (a) it should work for port 7, 8 and 9; and it should handle external triggers to arbitrary devices on the LAN. I will test that the whole firmware uploads in the coming days and, if so, release an alpha (A8) with this included for testing and feedback.

ntadmin, 2016-11-07:
As promised, 16A8 released: This is 16B7 with an alpha version of WOL for testing. Feedback welcome.

- https://github.com/ntadmin/DGND3700v2/releases/tag/NEWT16A8

ntadmin, 2017-09-04:
In the absence of negative feedback, I have assigned this to be part of the next beta release, which will be when #47 and #48 have been through a little bit of testing, expect them in NEWT17A1

vezza, 2017-09-11:
I'm still having problems... the router lan dhcp is set from 192.168.0.2  to 192.168.0.253, I leaved 192.168.0.254  outside the dhcp ip range in order to use it as wol address... the ruoter is unable to wol thr designed PC. If i try with a wol app on another device on the network il work fine... Sorry!

ntadmin, 2017-09-12:
I have a vague memory the .254 is also reserved for something on the router. But I may eb wrong. Could you try using a middle address that isn't used? If you go to Advanced > Administration > Attached devices it will tell you what numbers are in use! From memory, in a subnet, you should always avoid the first address, the last and the address before last (other than for a router) for general use.

vezza, 2017-09-12:
Today I have been able to do WoWan so this feature is working, thanks @ntadmin

ntadmin, 2017-09-13:
Fandabadozee :-)

closed 2017-09-13 by ntadmin