ntadmin/DGND3700v2 #40: Can't update V1.1.00.26_NEWT16B3 --> DGND3700v2_NEWT16B6.img

VincentYbarra, 2016-10-07:
What's the best way around this? 

Thanks.

ntadmin, 2016-10-08:
Hmmm, not experienced this one. Could you explain what does (and doesn't!) happen please?

A solution may be to 'update' to the stock Netgear firmware, and then to update from there to NEWTB6, but without further info it's hard to say.

ntadmin, 2016-10-10:
Further info: I don';t know why, but sometimes the updates stall in the middle of the upload. I have only experienced this since switching to using a separate modem for the VDSL and connecting to it from PPPoE port.

My solution is to unplug the PPPoE line, wait a minute or two and then do the update. However, if this fails, when the machine reboots, don't plug the PPPoE line back in. Simply try the update again and it then seems to work.

Any suggestions as to why this might be are welcome.

VincentYbarra, 2016-10-10:
Sorry forgot to update!

 Unplugging the PPPoE line fixed it!

closed 2016-10-10 by VincentYbarra

ntadmin, 2016-10-10:
Glad it worked for you!