ntadmin/DGND3700v2 #45: Not compatible with DGND3700Bv2

Just-Friend, 2017-05-14:
Please make it compatible with DGND3700Bv2 - it seems to be the same.
I am ready for experiment, because my DGND3700Bv2 don't work properly (I cannot reset it, after on/off new setting not applied) so maybe new firmware will fix it ;)

Here you can find all versions of the official firmware for DGND3700Bv2 (WW versions don't have even "B" in the description): http://downloadcenter.netgear.com/de/#searchResults

ntadmin, 2017-05-16:
Looking here http://wiki.openwrt.org/toh/netgear/dgnd3700, it the DGND3700Bv2 isn't listed. So, it is very hard to be clear on what is actually inside it.

- If the router is the same hardware but Germany branded (like the 3800B is for the 3700v1), then the firmware should work.
- If the router isn't, then the firmware stands a good chance of bricking the router.

So, you're options are:

1. Have a go with the 3700v2 firmware. Risk: bricked router.
2. Have a go with the 3700v1 firmware. Same risk

Or:

Have you tried a hard reset by doing a "debricking" as per https://github.com/ntadmin/DGND3700v2/wiki/FAQ%3A-Oooops-I%27ve-bricked-the-router using the stock firmware? This might get it into a working state which is a better place to start if you want to put custom firmware on anyway.

Just-Friend, 2017-05-19:
Thank you for the comment and sorry for the delay. I am just trying now to use my router "as is" and do not touch reset and On/Off button.
Anyway I am already tried to flash firmware through the TFTP, but my router just don't want to enter in firmware update mode. After all manipulations with reset button its starts only to flash Power on LED/Internet LED (red/green). I wanted to push in this router compatible old WW firmware and than update to modern/custom WW firmware.
I can also made a photo of this router for you, what is exactly inside of it.  But for this purpose I should wait when no one don't use it, take it off from the wall, disassemble it and make a couple photos.

ntadmin, 2017-05-19:
The delay is no problem - this is a hobby, and this is the first piece of custom firmware for a router I've written, so it may well take me some time to help, if I can help.

The key thing is to try and work out whether it is closer to a 3700v1 or a 3700v2. Looking at the firmware versions available:

- The netgear list: https://kb.netgear.com/2649/NETGEAR-Open-Source-Code-for-Programmers-GPL

- The link you give.

It does seem more likely to be a v2. So, what happens when you use the router's firmware update option to upload the latest WW firmware from Netgear for the 3700v2?

Just-Friend, 2017-05-22:
When I am trying to update firmware with 3700v2 firmware in the end of the update progress appears the message that this firmware for another model router.
I think, that if the only difference between 3700v2 and 3700Bv2 is the "B", than we need to remove model checking on your firmware or just to find out how remove "B" from the name of the DGND3700Bv2 router.

ntadmin, 2017-05-23:
I've done a quick comparison on one of the Netgear configuration files for the source code for the 3700v2 and what appears to be the source code for the 3700Bv2. The differences, ignoring SUB_VER which is simply a display label for the GUI.:

**3700v2**
```
BOARD_ID=DGND3700v2
HW_ID=ABL
VER=1.1.00.26
GUI_VER=1.1.00.18
HW_TYPE=DGND3700v2
ANNEX=A
FLASH=32M
BLOCKSIZE=64
```

**3700Bv2**
```
BOARD_ID=DGND3700Bv2
HW_ID=ADH
VER=1.1.00.21
GUI_VER=1.1.00.16
HW_TYPE=DGND3700v2
ANNEX=B
FLASH=128M
BLOCKSIZE=128
#------------------------------------------------#

```

- There are a couple of other minor variations which I think can be ignored.
- ANNEX refers to ADSL Annex. This may, or may not, be significant. If you are only using it as a router it probably doesn't matter.
- The HW_ID is different. I don't really know what this term refers to, but it may be significant.
- FLASH and BLOCKSIZE are very significant as they refer to how the image is constructed. In theory, as they are options, given the similarity of the code, changing them should be fine, but it is far from guaranteed.

So, this leaves us:
1.  Moderately sure that four changes to one of the firmware file and building from scratch will result in an image that will be accepted.
2. Not very convinced that this firmware will work, but it may not brick the thing - the ANNEX and HW_ID changes make me nervous.
3. Concerned that ADSL won't work in that image.

Given all of that, what are your views?

ntadmin, 2017-05-23:
https://github.com/ntadmin/DGND3700v2/releases/tag/NEWT16A8_B

Only for the brave.

Just-Friend, 2017-05-24:
Thanks a lot for this new firmmware, I am already tried to use it.
Firmware update through browser goes till the end with no errors, after that its automatically starts to update settings, auto-refreshes page and... nothing. Nothing changed in version info on routerlogin page. I've tried to update firmware through Firefox and Opera but with nothing changed despite the fact that firmware update progress goes with no errors till the end.
Anyway, thank you for the kind efforts ;)

ntadmin, 2017-05-24:
Positives:
 - your router isn't birdied :-)

Question: is the router connected to the internet at the time? If so, how?

That may help clarify how to get firmware on to it.

Just-Friend, 2017-05-25:
Yes.
It is connected with WiFi and I am just tried to do that with reboot and disconnected internet. But firmware is not accepted by router.

ntadmin, 2017-05-25:
Does it give you a failure message or just not work?
Can you describe the sequence of lights as you go through please?

Just-Friend, 2017-05-31:
Sorry for a long delay, ntadmin.

There is no failure message its just "downloading firmware in router" like in emulation mode and happens nothing :(
Sequence at start: 1LED(On/Off): Green. 2LED: No. 3LED: No. 4LED: Green(WiFi2.4). 5LED(WiFi5): Blue. 6LED: Green. 7LED(Internet): Green. 8LED(USB): Green. 9LED: No. 10LED(WPS?): Green.
Sequence at about 50% downloaded firmware: 1LED(On/Off): Green. 2LED: No. 3LED: No. 4LED: Green(WiFi2.4). 5LED(WiFi5): No. 6LED: Green. 7LED(Internet): Orange. 8LED(USB): Green. 9LED: No. 10LED(WPS?): Green.
Sequence at about 60% downloaded firmware: 1LED(On/Off): Green. 2LED: No. 3LED: No. 4LED: Green(WiFi2.4). 5LED(WiFi5): Blue. 6LED: Green. 7LED(Internet): Green. 8LED(USB): Green. 9LED: No. 10LED(WPS?): Green.

So, about on 50% Internet LED became Orange and disappears 5Hz WiFi, but only on a shot time, after "upgrading firmware" and "updating setting" nothing happens. Router even doesn't rebooting in this process. When I force reboot from routerlogin page or by button nothing changes in version and description of the modemrouter, appears my old setting which I have used for years before, now I have another settings for internet and should manual change this setting (option "Backup/Restore Settings" doesn't work for me).
Hope it helps to "diagnose" the problem ;)

ntadmin, 2017-06-03:
Please don't worry about the delay - I can only get to this every few days anyway.

Is it exactly the same sequence if you try to update to an official firmware?

Suppose you try to update to an older version of the netgear firmware, what happens?

This is looking like a problem beyond my experience, but I'll keep helping as much as I can!

Just-Friend, 2017-06-08:
If I am trying to updated to an official firmware the sequence is: Power LED (blinks green and than light on red), all other LED don't work and don't blink. After complete all LEDs starts work like usual.
I can rollback to an old V1.1.00.17 firmware but not to v1.1.00.08 or v1.1.00.12.

By the way, I have tried to update to your firmware (Alpha release for 3700Bv2) from another PC with Internet Explorer and due this process appears falure message about not capable firmware. So, maybe you can try to Find/Replace modem name from DGND3700v2 to DGND3700Bv2, because in some way router detects that firmware is from another model.

ntadmin, 2017-06-26:
Just so you know, I am working on this, slowly, in the background. I've identified at least one issue, but it's probing a bit hard to make it work, but I hope to have a new binary for you to try when I have cracked it!

ntadmin, 2017-06-26:
One issue: https://github.com/ntadmin/DGND3700v2/issues/46

ntadmin, 2017-06-26:
OK, here we go: https://github.com/ntadmin/DGND3700v2/releases/tag/NEWT17A1_B

As usual, I make no promises. It may brick things, but it compiled better than last time :-)