ntadmin/DGND3700v2 #48: Cant forward more then a single port

Shad0wGeneral, 2017-08-20:
Ok so as the title says I'm trying to forward more then one port but every time I try to it just updates the first one it doesn't add a new forwarded port to the list. [https://i.gyazo.com/d42c2abd908cea692d885736238ba7c9.png](url)

ntadmin, 2017-08-20:
Hi @Shad0wGeneral, I'm away from home for the next week but will look into this when I am back at home. Thanks.

ntadmin, 2017-08-28:
Ok, I've had a quick look at it and I can reproduce it. There was a similar issue previously, so I will see if that provides any insight.

ntadmin, 2017-08-28:
Exploring whether this has any links to bug #35, commits 7008037737d5fd483d1ccb51d6e72cea54d364f5 and fde8b92114dce02654c9cd08728190121793ef60

ntadmin, 2017-08-29:
So, FW_forward calls for rule_in:
- If called with 'edit' it calls for rule_in.htm with `todo` set to 'edit', `ruleSelect` set to the index for the row to edit and this works.
- If called with 'add' it calls for rule_in.htm with `todo` set to 'newfile'.
In both cases our `render_page` attempts to render rather than passing it off to the netgear page renderer.

ntadmin, 2017-08-29:
So, adding:
`this.form.h_ruleSelect.value = document.getElementsByName('in_sel').length`
to the onClick for Add before `stdAction` forces it to provide a correct ruleSelect value.

However, while we are here, doesn't doesn't seem to work, so that's next ...

ntadmin, 2017-09-03:
OK, I think that the version in github on the master (only!) branch solves this, but not quite releasing it yet. If an active tester is out there I can throw an alpha up.

ntadmin, 2017-09-13:
This is now working as far as I can see so I will close this issue. Future problems should be openede as new issues please.

closed 2017-09-13 by ntadmin