ntadmin/DGND3700v2 #51: USB mount system doesn't put devices into approved list.

ntadmin, 2017-09-14:


ntadmin, 2017-09-14:
@vezza - I think I know what the issue is here, but it will take a bit of work. I'm pretty much in solid meetings from now until the end of Saturday and then I'm on the road for six days, but I will get onto it over time.

ntadmin, 2017-09-14:
@vezza - there's also the background issue that the system I have been using to build the firmware is no longer supported by the distribution as it is so old, and this may slow things down. It's also worth noting that, in the longer run, this will make it increasingly hard to maintain the firmware.

vezza, 2017-09-16:
@ntadmin - I never tried to build the firmware miself but I'm interested in it... what distribution is needed to build it?

ntadmin, 2017-09-17:
@vezza good question. You have to set up a machine (or a virtual one) running a 32 but version of Linux. I'll look out some links for you.

ntadmin, 2017-09-17:
@vezza I think that the post https://community.netgear.com/t5/DSL-Modems-Routers/Customising-the-firmware-for-the-DGND3700v2/m-p/1074375#M9863 has links in it to how to get things going. I think my other posts on hay forum also make most things clear. 

Happy to answer questions as you give it a go, I'm on the road until next weekend, I'll help if possible.

Ideally we need to make it so that the firmware can be compiled on an modern Linux distribution :-)