ntadmin/DGND3700v2 #53: Port Forwarding not working on NEWT17A2

vezza, 2017-09-22:
I am trying to forward all incoming connection to a LAN ip address... is there a way to test port forwarding from the router's ssh shell?

ntadmin, 2017-09-23:
The way I test the port forwarding is:
1. Get the IP address we are running on (shown on advanced home)
2. Attempt to connect to that address (using the right protocol for the right port) from a computer on my network.
So, if my IP address as seen by the world was 100.100.100.100 and I was interested in running an ssh shell via port 678, I would attempt to make an ssh connection to 100.100.100.100:678 and see what happened ...

vezza, 2017-09-27:
tomorrow I will re-test the router... will report back. Good night!

ntadmin, 2017-10-01:
Is this now working as well?

vezza, 2017-10-01:
Yes... I've just finished some testing now... It works fine

ntadmin, 2017-10-02:
Cool, thanks for confirming.

closed 2017-10-02 by ntadmin

vezza, 2017-10-04:
Unfortunately I have to report that I'm still experiencing network problems under certain config, in particular on wi-fi connected devices, they cannot connect to internet,so I've tried to manually restart WLAN:

```
# /bin/rc_app/rc_wlan restart
killall: wps_ap: no process killed
sh: can't create /proc/sc_led/wifi_mode: nonexistent directory
sh: can't create /proc/sc_led/wps_running: nonexistent directory
Setting SSID: "NETGEAR32"
Setting SSID: "NETGEAR-Guest"
Setting SSID: "NETGEAR-Guest2"
Setting SSID: "NETGEAR-Guest3"
device wl0.1 is not a slave of group1
device wl0.2 is not a slave of group1
device wl0.3 is not a slave of group1
Bus error
Setting SSID: "NETGEAR32-5G"
Setting SSID: "NETGEAR-5G-Guest"
Setting SSID: "NETGEAR-5G-Guest2"
Setting SSID: "NETGEAR-5G-Guest3"
device wl1 is not a slave of group1
device wl1.1 is not a slave of group1
device wl1.2 is not a slave of group1
device wl1.3 is not a slave of group1
Bus error
lan_ifnames=eth0 wl0 wl1 wl0.1 wl0.2 wl0.3 wl1.1 wl1.2 wl1.3 
lan1_ifnames=
========idx=0
Setting SSID: "NETGEAR32"
Setting SSID: "NETGEAR-Guest"
Setting SSID: "NETGEAR-Guest2"
Setting SSID: "NETGEAR-Guest3"
Chanspec set to 0x2d08
/sbin/wl: Not Permitted
/sbin/wl: Undefined error
sh: /bin/igd: not found
setting oob: lan_wps_oob 
wps_cfg_state = 1, wps_config=ap-er
wscLedSet: led_action 0, led_blink_type 0, led_event 0, led_status 0
wscLedSet: led_action 0, led_blink_type 0, led_event 0, led_status 0
killall: wps_det: no process killed
issueServiceCmd
Bus error
setupAll end
========idx=1
```
Don't know if the above output is useful, anyway it seems to me like a dns or routing table problem related to port forwarding... 
I can bypass the issue disabling port forwarding rules and enabling DMZ to my web server.
I must say that I've not deeply tested this issue. I'll try to investigate asap...

ntadmin, 2017-10-05:
I guess that the quick test would be to remove all port forwarding and see if that makes everything else work.

If it does, then we know the issue is with the port forwarding. If not, the issue is elsewhere.

If it is the forwarding, then it may be that one of your rules is triggering unexpected behaviour, if so we need to identify which one!