ntadmin/DGND3700v2 #54: Wi-Fi and DSL off after reboot NEWT17A2

vezza, 2017-09-22:
As I am experiencing some problems with port forwarding (see separate issue) I tried to reboot the router. After reboot waited some time but DSL and Wi-Fi remains off...

ntadmin, 2017-09-23:
Very odd. Quick question: you did reconnect the DSL line after the reboot right? I know that sounds stupid, but I've made that mistake before. Not that it explains the Wifi being off, but it's a place to start!

ntadmin, 2017-09-23:
Then, have the wifi setting changed when you view them from the router's web interface?

vezza, 2017-09-27:
with "you did reconnect the DSL line after the reboot right" do you mean:
1) I plugged the DSL line cable back to the router;
or
2) I clicked the connect buttun in the DSL settings router's page
If you mean 2) I waited almost 5 minutes after reboot but the DSL autoconnection did not take place (I have configured the router so that It auto-reconnect when DSL line is available...
If you mean 1) I did not unplugged the DSL cable before rebooting the router...
Anyway, tomorrow I will re-test the router... will report back. Good night!

ntadmin, 2017-09-28:
Very odd - sorry about this, but I'm a bit bemused as I haven't altered the DSL things for a long time. My modem is now running PPPoE, and I am unable to test the DSL side anymore.
If it still fails when you next try, could you 'upgrade' the firmware to the previous firmware and see if that helps? Thanks,

vezza, 2017-10-01:
Well I've re-tested... Now everything is working fine... probably I did something wrong or didn't waited the necessery time, sorry for the false bug report and time waisted

ntadmin, 2017-10-01:
No worries, just glad it's working now.

closed 2017-10-01 by ntadmin