ntadmin/DGND3700v2 #55: Can't get an IP address after switching to AP mode

nickmt, 2018-01-05:
Initially when upgrading to the custom firmware, my DGND3700v2 was in AP mode - this was to extend the wifi of my main modem/router, following these guides:
https://forums.whirlpool.net.au/archive/2550688#r51167902
https://community.netgear.com/t5/DSL-Modems-Routers/HOW-TO-Extend-Your-WiFi-Using-DGND3700-Routers/td-p/1005524

However after doing this, my I could no longer get an IP from the main modem/router. I was able to reset it and access it on its own, but as soon as I connected it back up and switched to AP mode again, I lost the IP again.

Even though it doesn't have an IP address, it still seems to function correctly; ie: I can still connect to the access point or through ethernet and hit my main modem/router. It just means I can't access the DGND3700v2's web interface at all and I need to hit the factory reset button to be able to do anything with it again.

Thanks for all the work you've put into this!

ntadmin, 2018-01-05:
Thanks for trying out the firmware. I currently use my 3700v2 as a wireless base, but I don't bother with the AP (wireless repeater) functions. A couple of questions:

1. Did it work as you would expect with the original firmware?
2. When you look at the router providing DHCP, what address is given to the DGND3700v2? It may well be different from the one you assigned in the static stage of the instructions?
3. What happens when you browse to that address?

Thanks,

nickmt, 2018-01-15:
Sorry about the delayed response.

1. Yeah, it appears to work fine on the original firmware.
2 & 3. No address appears to be given to the DGND3700v2, even though I've given it static address. For this, I'm using an ISP-provided Huawei HG659. On that end, it comes up with the MAC address, but no IP associated with it.

ntadmin, 2018-01-15:
No worries about the delay. Another question, I'm afraid.
 - When the router is running in AP mode but you can't see it on it's ip address, are you connecting through:
 - the WAN port
 - a LAN port
 - its wireless?

Out of interest, is there any reason to run in AP mode and not just leave it running as a router with the other services disabled?

nickmt, 2018-01-21:
I'm using the WAN port.

I couldn't get it working with disabling the other services - it seemed it had to be on AP mode for it to work. The Whirlpool thread I linked in the first post effectively sums up the issues I had with it.

I can do a factory reset in a couple of days and start from scratch, feel free to send through anything you want me to try.

ntadmin, 2018-01-23:
Ok, that’s helpful to know.
Looking at the Whirlpool thread, I can’t quite see the issues you refer to: in essence it seems to say ‘I got it working by using AP mode and the WAN port.

One things to check: is the IP address listed on the AP enable pages the same as the router’s IP as set statically? If not, it will need to be set to be the same on the AP page.

I wouldn’t worry about doing a system reset yet.

ntadmin, 2018-01-23:
Looking at it another way, I’m surprised that you haven’t managed to get it working via a LAN port: that’s how I use mine. In essence, I disable everything that can be disabled (DHCP, DNS, etc etc) and only allow the wireless to run. On the DHCP page there may be an ‘enable DHCP pass through’ option which should be enabled. I’m away from home so can’t be more specific, sorry.

nickmt, 2018-02-17:
I was able to get it working by doing the same thing, thanks. I did have to reset it though since I still wasn't getting an IP, but now it's working fine. As for the DHCP I just had to disable it and it was fine.

ntadmin, 2018-02-17:
Glad it's working for you now!
We should from an NT club!

closed 2018-02-17 by ntadmin