ntadmin/DGND3700v2 #56: Wireless Bridge support

deevus, 2018-02-07:
Is there any way to get wireless bridging working? Wireless repeating requires disabling security which I don't want to do.

Forest-Feighner, 2018-02-09:
I've been thing about working on wireless bridging as well. My thought was that it could just be set up via wl.

I just noticed your post as I stoped by to mention to ntadmin that I noticed some flashing option for wl. I don't currently have access to the router but it's on my todo list.

ntadmin, 2018-02-09:
I’m pretty sure it should be possible, and interesting to hear about the options for flashing.

Sadly, I’m not in a position to do much to the router anymore but very happy for others to go for it. Please do add to the distribution, I’m really happy for people to fork it and add to it and I can merge back in and release.

ntadmin, 2018-02-09:
Sorry about the accidental close and reopen there.

Ivoz, 2018-02-16:
Yes, it's quite sad that I have this older router sitting here, but it's only capable of wireless bridging with WEP security, which may as well be open for all the security it gives nowadays.

ntadmin, 2018-02-16:
Well, it does make a good wireless base if you can get a wireless to it!