ntadmin/DGND3700v2 #58: Unable to load new firmware

maxrelic, 2018-02-24:
After loading ver. alpha 17A2 the DGND3700v2 refuses to load any other firmware, it seems the flash is write protected.  I am still able to backup/restore settings.

ntadmin, 2018-02-24:
If you have an ADSL or VDSL connection, please disconnect the wire and reboot before trying. This is a known issues with the firmware, sorry.

maxrelic, 2018-02-24:
thank you netadmin, it didn't work even after a power reboot and line disconnected. Eventually I used nmrpflash to install the Netgear's 1.00.26.  BTW looking ahead for your final release, keep the good work going.

ntadmin, 2018-02-24:
Sorry that didn’t work for you. I should probably note here that I am no longer able to work actively on this project as I am now away from home too much, and the router is also no longer in use (other than as a WiFi base) at my home. So, I welcome all contributions to the code, but I am unable to do more work on it myself, I’m afraid. Please do download and seek to improve :-)

closed 2026-11-16 by maxrelic