ntadmin/DGND3700v2 #59: Dynamic DNS not working

HackerWolf, 2018-03-14:
Running NEWT17A2.
Every dyndns service seems not to work. Tried everything. If this project is not died, I'll be happy to share details to try a fix,

thanks

ntadmin, 2018-03-18:
Please do share the details - it might be easy to solve in which case we’ll see. I no longer use mine, so I can only make theoretical suggestions.

Anyone is welcome to download the code and propose improvements!

timglembin, 2018-04-14:
The Makefile for packages downloads the latest version of [inadyn-mt](https://sourceforge.net/projects/inadyn-mt/files/latest/download), so you probably need to download the latest version of [servers_additional.cfg](https://sourceforge.net/projects/inadyn-mt/files/servers_add/servers_additional.cfg/download) as well. I don't think the project version is the latest.

Once I did this, DDNS works as expected. I was able to add DNSdynamic.org quite easily as well.

HackerWolf, 2018-04-16:
@timglembin do you mean a new from-scratch build of CFW or only to manually update that file over a flashed device? I'm trying to compile this, I succesfully built kernel and source, but struggling on extrabin and I can't go on.

ntadmin, 2018-04-17:
I think that additional_Servers should probably be downloaded by the Makefile when it downloads the inadyn-mt source and then put in the right place. This would be in the 'packages' phase of the build - that's the Makefile to update there.

@HackerWolf - what is failing on the extrabin build? Although I'm not in a position to build myself, if you post the error (maybe start a separate issue) I'll see if the error triggers any ideas!

timglembin, 2018-04-21:
@HackerWolf I updated the additional_servers.cfg file manually. The project version won't affect compiling, it's a runtime issue.

If you're having problems with compiling the packages, try downloading the source for each individual package manually, & manually creating the folders. The specifics you need are in the packages Makefile.