ntadmin/DGND3700v2 #60: Adblock not allowing pages clicked from google search results

timglembin, 2018-05-14:
Pages would not load when clicked from google search results.

I had to add www.googleadservices.com to the allowed adblock sites to get around this.

ntadmin, 2018-05-14:
Good to hear that the ‘allow’ function is working.

The first results that google gives you are ‘sponsored’ results, ie they are adverts honed to your search terms. If you look at them you will see a little market next to them telling you that they are an advert.

The behaviour you describe is how it is designed to work: when ad blocking is enabled all known advert sites (including Google) are blocked, it is up to the user to allow them.

It would theoretically be possible to add a series of yes/no options which allowed certain known advertiser through, such as the one you give. If anyone would like to add that to the system, please go for it!

timglembin, 2018-05-14:
Oh yes for sure. Sorry I suppose I meant this more as an FYI than an issue, in case anyone else noticed it & didn't connect the two.

I'm fairly certain Google has made some changes as I'd only starting seeing this recently, & not on adverts but on actual results items. Hence why I needed to add the allow rule. It doesn't seem to happen often though..

Or maybe I've just picked up a virus 😢

timglembin, 2018-05-14:
Is googleadservices.com even owned by Google? I think I'd better do a little research..