ntadmin/DGND3700v2 #61: New feature: openvpn

vezza, 2018-07-19:
Do you plan to add support for openvpn in the future?

ntadmin, 2018-07-20:
That's a great idea, but I'm longer able to work on the code for this project. You are most welcom to contribute improvements via the usual github means and I will approve them etc.
If you would like to take on the project, please let me know!

vezza, 2018-07-22:
Not sure if I'll be able to manage this project as I'm not a developer... anyway I may try. I feel more comfortable if a developer can review the commits as I don't fully understand all the code. On the other hand it is always possible to revert the commits so that encourage me to try to manage the project. At the moment I still haven't understood how you build the binaries... I can't find any docs explaining how to do a release starting from source code...

ntadmin, 2018-08-02:
@vezza Sorry it's taken so long to reply.

Two community posts help a lot with getting going:

https://community.netgear.com/t5/DSL-Modems-Routers/Compiling-the-toolchain-for-the-DGND3700v2/m-p/1073898

https://community.netgear.com/t5/DSL-Modems-Routers/Compiling-the-stock-firmware-for-the-DGND3700v2/td-p/1074374

Some of the later steps in the second process are in the Makefile, but none of the first.

Hope that helps,

vezza, 2018-08-23:
I've been able to build the toolchain but now I'm not able to build the firmware, I'm getting this error:
/bin/ld: i386 architecture of input file `crc32.o' is incompatible with i386:x86-64 output
what I'm doing wrong?

ntadmin, 2018-08-24:
I’m working away from home for the next ten days, but I will look at the issue as soon as I can on my return. What Virtual environment are you using the build in?

    Neil

vezza, 2018-08-24:
I set up a Linux container with Centos 6 x86_64 on top of a Fedora 28  x86_64 host. The toolchain is in: /opt/toolchains/uclibc-crosstools-gcc-4.4.2-1-with-ftw
and:
`# ln -s /opt/toolchains/uclibc-crosstools-gcc-4.4.2-1-with-ftw /opt/toolchains/uclibc-crosstools-gcc-4.4.2-1`
the kernel is in: /opt/toolchains/DGND3700v2_V1.1.00.26_WW_src_bak/
when I enter the:
`# make kernel`
I get the error:
/bin/ld: i386 architecture of input file `crc32.o' is incompatible with i386:x86-64 output

vezza, 2018-08-24:
problem solved... I was missing libstdc++.i686 package. Now I'm facing another problem doing:
`# make source`
I'm getting a lot of messageslike this:
tar: target/dev/tty: funzione "mknod" non riuscita: Operation not permitted
..
tar: target/usr/dev/bcmadsl1: funzione "mknod" non riuscita: Operation not permitted
tar: Uscita con stato di fallimento in base agli errori precedenti
make[1]: *** [pre_fs] Error 2
make: *** [source] Error 2

vezza, 2018-08-25:
Don't  know what happened but I restarted lxc.service end the container and now I am able to get beyond...
unfortunately now the `# make source` stops with the following error:
`Making all in libfuse-lite
 /usr/bin/mips-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I. -I.. -I../include/fuse-lite -g -O2 -Wall -MT libfuse_lite_la-fuse.lo -MD -MP -MF .deps/libfuse_lite_la-fuse.Tpo -c fuse.c  -fPIC -DPIC -o .libs/libfuse_lite_la-fuse.o
../libtool: line 1322: /usr/bin/mips-linux-uclibc-gcc: No such file or directory
make[5]: *** [libfuse_lite_la-fuse.lo] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [nvram] Error 1
make[1]: *** [apps] Error 2
make: *** [source] Error 2`

vezza, 2018-08-25:
keep working on that...
`ln -s /opt/toolchains/uclibc-crosstools-gcc-4.4.2-1/usr/bin/mips-linux-uclibc-gcc /usr/bin/mips-linux-uclibc-gcc`
bypassed the problem and building is now proceeding... anyway stopped again by:
apps: make radvd-1.1 
/bin/sh: /work/Platform/DG834GV_26/code/projects/DG834_V4/Neutral2/Source/apps/radvd-1.1/missing: No such file or directory
unfortunately installing radvd package doesn't make any difference so I'm now stuck on that...

vezza, 2018-08-25:
It seems that glibc-headers package is missing but, after installing it, the command `# make source` gives the same output:
apps: make radvd-1.1 
In file included from log.c:17:
./includes.h:99:22: error: ifaddrs.h: No such file or directory

ntadmin, 2019-04-08:
Hi, sorry, been away from this stuff for a while. Are you compiling using a 64bit or 32bit host for compilation, I have a vague memory that, for some reason, the 64 bit host cross compiler fails in this way, very bizarre. The only solution I found was to create a 32 bit host and cross compile from there.