ntadmin/DGND3700v2 #63: Can I get compiled, ready to install files?

Blum, 2019-04-05:
Hello,
thank you for your work.
Can I get compiled, ready to install files? Or at least some brief instructions how to compile it myself?

Cheers

ntadmin, 2019-04-08:
Hi Blum,
Before going further, you should read the disclaimers and notes in the readme file here: https://github.com/ntadmin/DGND3700v2

And be aware of the issue mentioned here: https://github.com/ntadmin/DGND3700v2/issues/62

Ready to install images are here: https://github.com/ntadmin/DGND3700v2/releases

If you are interested in compiling, you should read the thread here: https://github.com/ntadmin/DGND3700v2/issues/61

Blum, 2019-04-09:
Thank you, 
Think I'll take the chance, your version has some functionality I certainly need. Fingers crossed.